PowerBroker2/SafeString #5: Program space

patfelst, 2020-07-10:
Hi again PB, this looks really interesting. I had a look at the source code, and it's pretty huge. Do you know how much program space including it uses? Or is there some smarts applied by the compiler/linker to only include the methods it uses?

I'm using a few C strings (char arrays) in my current development on an ESP32, and need to remove a sub-string from a source string, and some other string manipulation, this library would make it easy, but not sure it's worth it when the library is so huge. I recognise there are other benefits too like avoiding heap fragmentation and speed etc.

Thanks in advance!

PowerBroker2, 2020-07-10:
I actually didn't write the code, so I'm not familiar enough to answer your questions. Test compiling might answer some of your questions. From what I can tell, SafeStrings are basically char array buffers with backwards compatible String methods. You might be able to contact the original author by posting an issue [here](https://github.com/drmpf/SafeString).

closed 2020-07-10 by PowerBroker2

drmpf, 2020-07-10:
Hi, You can contact me (the author) via pfod.com.au at support.
There is a define near the top the SafeString.h
// to remove all the error messages, comment out
#define SSTRING_DEBUG
// this saves program bytes and the ram used by the SafeString object names
which will save space after you have you program running.
Compilers now are usually aggressive in removing unused code also

patfelst, 2020-07-11:
thanks, when I get a chance, I'll test compile it in my project and see how much space it uses.