PowerBroker2/SafeString #28: Arduino Uno throws errors on Verify; no members named 'flush' or 'avaiiableForWrite'

joelmaxuel, 2021-02-12:
Attempting to compile for the Arduino Uno (particularly with a Maker Uno 1.1) but get the following error:

```
/home/joel/Build/static/sketchbook/libraries/SafeString/SafeString.h: In member function ‘void SafeString::DebugPrint::flush()’:
/home/joel/Build/static/sketchbook/libraries/SafeString/SafeString.h:824:25: error: ‘class Print’ has no member named ‘flush’
          currentOutput->flush();
```

In tracking down, it seems the Uno also has no flush in Print

```
#if defined(ESP_PLATFORM)
    	// ESP32 has no flush in Print!! but ESP8266 has
#else
        	currentOutput->flush();
#endif
```

Removed the above ifdef however there are additional errors arising in other files (also wrapped in ESP_PLATFORM ifdefs).

```
/home/joel/Build/static/sketchbook/libraries/SafeString/BufferedOutput.cpp: In member function ‘void BufferedOutput::connect(HardwareSerial&)’:
/home/joel/Build/static/sketchbook/libraries/SafeString/BufferedOutput.cpp:92:26: error: ‘class HardwareSerial’ has no member named ‘availableForWrite’
   int avail = serialPtr->availableForWrite();
                          ^
/home/joel/Build/static/sketchbook/libraries/SafeString/BufferedOutput.cpp: In member function ‘void BufferedOutput::connect(Stream&, uint32_t)’:
/home/joel/Build/static/sketchbook/libraries/SafeString/BufferedOutput.cpp:152:28: error: ‘class Stream’ has no member named ‘availableForWrite’
     int avail = streamPtr->availableForWrite();
                            ^
/home/joel/Build/static/sketchbook/libraries/SafeString/BufferedOutput.cpp: In member function ‘int BufferedOutput::internalStreamAvailableForWrite()’:
/home/joel/Build/static/sketchbook/libraries/SafeString/BufferedOutput.cpp:283:24: error: ‘class HardwareSerial’ has no member named ‘availableForWrite’
     avail = serialPtr->availableForWrite();
                        ^
/home/joel/Build/static/sketchbook/libraries/SafeString/BufferedOutput.cpp:288:24: error: ‘class Stream’ has no member named ‘availableForWrite’
     avail = streamPtr->availableForWrite();
                        ^
```

drmpf, 2021-02-12:
V3.0.6 has different code to what you have posted.
Try updating to V3.0.6 and then if you still have errors enable File->Preferences,  Verbose output for Compile and include a compile line so we can see what is defined.

joelmaxuel, 2021-02-13:
Same error. Seems as though the Uno does not apply to any of the extra guards.

```
#if defined(ESP_PLATFORM) || defined(ARDUINO_SAM_DUE) || defined(ARDUINO_ARCH_STM32F1) || defined(ARDUINO_ARCH_STM32F4)
        // ESP32 has no flush in Print!! but ESP8266 has
#else
                currentOutput->flush();
#endif

```

drmpf, 2021-02-13:
Please email pfod.com.au via the contacts link with your sketch and complete compiler output.  Uno works fine for me here
Or post here if you prefer.

joelmaxuel, 2021-02-13:
Just looked at the version of my Ardiuno IDE, even when using Debian Testing it is painfully old. Grabbed a copy from FlatPak and I had no issues compiling. Apologies for the intrusion.

closed 2021-02-13 by joelmaxuel