PowerBroker2/SafeString #40: SafeString 4.1.5 causing SSD1306 OLED failure to allocate

tradJack, 2021-05-29:
I have been using safestring successfully for over a year.. however lately, I have gotten bogged down with strange interactions when using SafeString AND Adafruit SSD1306 OLED 128x64 display..

attached is the demo file from the adafruit library with simply safestring included and if u uncomment the several createSafeStrings, it causes the display to not allocate..

??????

regards-- jack
[ssd1306_128x64_i2c_SafeString.zip](https://github.com/PowerBroker2/SafeString/files/6564718/ssd1306_128x64_i2c_SafeString.zip)

PowerBroker2, 2021-05-29:
@drmpf

drmpf, 2021-05-29:
**Could not reproduce issue**
Installed Adafruit GFX Library version=1.10.9
Adafruit BusIO version=1.7.3
Adafruit SSD1306 version=2.4.5
SafeString version=4.1.5

Modified ssd1306_128x64_i2c_SafeString.ino setup to
```
void setup() {
  Serial.begin(9600);
  for (int i = 10; i > 0; i--) {
    Serial.print(i); Serial.print(' '); delay(500);
  }
  Serial.println();  
  SafeString::setOutput(Serial);
  if(!display.begin(SSD1306_SWITCHCAPVCC, SCREEN_ADDRESS)) {
    Serial.println(F("SSD1306 allocation failed"));
    for(;;); // Don't proceed, loop forever
  }
  Serial.println(F("Passed allocation"));
```
Output was
```
10 9 8 7 6 5 4 3 2 1 
Passed allocation
```

tradJack, 2021-05-29:
thats not the program I attached..

when I 'un-comment' the createSafeString variables, then the program
crashes..
will fail to allocate..
even though the variables arent even used for anything..!!

drmpf, 2021-05-30:
**Under investigation**
Can reproduce on Uno when modified supplied program to un-comment
```
createSafeString(tmpStr,50);
createSafeString(captureData,10);
createSafeString(camStr,4);
createSafeString(vgStr,4);
createSafeString(volStr,4);
createSafeString(field,10);
```

tradJack, 2021-05-30:
thankyou sir

tradJack, 2021-05-30:
I had the problem using mini pro..

drmpf, 2021-05-30:
Immediate work around is to replace the global SafeStrings with char[  ]
```
char tmpStr_c[50 + 1];
char captureData_c[10 + 1];
char camStr_c[4 + 1];
char vgStr_c[4 + 1];
char volStr_c[4 + 1];
char field_c[10 + 1];
```
Then at the top off the loop() add
```
  cSFA(tmpStr, tmpStr_c);
  cSFA(captureData, captureData_c);
  cSFA(camStr, camStr_c);
  cSFA(vgStr, vgStr_c);
  cSFA(volStr, volStr_c);
  cSFA(field, field_c);
```
to work with them as SafeStrings

tradJack, 2021-05-30:
I already have rewritten my serial_receice routine from safestrings to
char[]..char arrays and safestring are nearly same thing.. and it looks to
be stable now--

regards--jack

bool serial_Rec(){static char inputStr[]={' ',' ',' ',' ',' ',' ',' ',' ','
',' ',' ',' ',' ',' ',' ',' ',' ',' ',' ',' '};
                     //bool data_recd=false;
                  char inChar=' ';
                  int n=0;

                   n=0;
                   if(Serial.available()){while(Serial.available()){// get
the new byte:

inChar = (char)Serial.read();

delay(13);
                                                                      //
add it to the inputString:

inputStr[n]= inChar;
                                                                      n++;

//Serial.print(inChar);

if(inChar=='\n')break;

if(n>=15)break;
                                                                      }

                     if(n>=2){//format= cam:xx,vg,carosel
                              //delay(100);
                              //Serial.print("inputStr=");
                              //for (int x=0; x<=n+1; x++){delay(100);
                              //
 Serial.print(inputStr[x]);}
                             // Serial.println();

if(inputStr[0]=='c'){if(inputStr[1]=='a'){if(inputStr[2]=='m'){if(inputStr[3]==':'){
                                 //Serial.println("start");
                                 n=4;
                                 camStr=inputStr[4];
                                 if(inputStr[5]!=','){n=5;
                                                      camStr+=inputStr[5];}

//Serial.println(camStr);
                                                      n++;

//Serial.print("n=");Serial.print(n);Serial.println(inputStr[n]);

if(inputStr[n]==','){n++;

//Serial.println("kk");

 vg=inputStr[n];

//Serial.println(vgStr);

 n++;

 if(inputStr[n]==','){n++;

                    carosel=inputStr[n];

                    captureStr="cam";

                    //Serial.println("dd");

                    return true;}}}}}}}

if(inputStr[0]=='v'){if(inputStr[1]=='g'){if(inputStr[2]==':'){{

//Serial.print("n=");Serial.print(n);Serial.println(inputStr[n]);
                              vg=inputStr[3];
                              //Serial.println(vgStr);
                              captureStr="vg";
                              //Serial.println("dd");
                              return true;}}}}

 if(inputStr[0]=='v'){if(inputStr[1]=='o'){if(inputStr[2]=='l'){if(inputStr[3]==':'){{{

//Serial.print("n=");Serial.print(n);Serial.println(inputStr[n]);
                              volStr=inputStr[4];
                              volStr+=inputStr[5];
                              captureStr="vol";
                              //Serial.println("dd");
                              return true;}}}}}}

                     }
                     return false;
}

drmpf, 2021-05-30:
**Close Issue as no error.**

Required memory exceeds the available RAM.
The display requires 1024 bytes
With the global SafeStrings defined and with the local variables in setup() the remaining memory is ~1081 on UNO.
malloc on Arduino AVR needs free space of 128 to succeed.
1024 + 128 > 1081 so the allocation fails.
commenting out the 
createSafeString(tmpStr,50);
allows the allocation to succeed. 
With SafeString #define SSTRING_DEBUG (the default)
One SafeString takes about 53bytes + length of the SafeString variable name + length
Subsequent SafeStrings take about 30bytes + length of the SafeString variable name + length
so removing the createSafeString(tmpStr,50); statement
frees up about 80 bytes (since there are still other SafeStrings defined) which enough to allow the display allocation to succeed.

tradJack, 2021-05-30:
thankyou sir for the detailed explanation..I was somewhat aware of the high
need of SRAM for the SSD1306 display but I didnt realize how closely to
all-used-up I am running regardless if I use safestrings or not.. I will
now look to minimize my variables and trim those I have to have..
very much appreciate--
regards,
jack in arkansas

drmpf, 2023-07-14:
@PowerBroker2 This can be closed. Thanks

closed 2023-07-14 by PowerBroker2