PowerBroker2/SafeString #44: Idea: create a function like String, is possible?

Attigliuzzo, 2021-07-22:
Hi. I'm a noob programmer and my question is, we know i can write something like this:

```
String sendData(String command, const int timeout, boolean debug)
{
    String response = "";  //initialize a String variable named "response". we will use it later.
    
    Serial.print(command);   //send the AT command to the esp8266 (from ARDUINO to ESP8266).
    
    long int time = millis(); //get the operating time at this specific moment and save it inside the "time" variable.
    
    while( (time+timeout) > millis()) //excute only whitin 1 second.
    {      
      while(Serial.available())   //is there any response came from the ESP8266 and saved in the Arduino input buffer?
      {
        char c = Serial.read(); //if yes, read the next character from the input buffer and save it in the "response" String variable.
        response+=c; //append the next character to the response variabl. at the end we will get a string(array of characters) contains the response.
      }  
    }    
    
    if(debug)  //if the "debug" variable value is TRUE, print the response on the Serial monitor.
    {
      Serial.print(response);
    }    
    return response;  //return the String response.
}
```
Can it be possible something similar but using safestring? 
i mean eg. "safestring sendData(........){......}"
Thanks

This code is used with an esp8266-01 and i want to understand if the failure of my system to run for days is due of String overflow.

drmpf, 2021-07-23:
Perhaps the failure is due to String OR just problems with the underlying libraries, see 
[Out-Of-Memory on ESP32 and ESP8266 – Add Periodic Automatic Reboots](https://www.forward.com.au/pfod/ArduinoProgramming/ArduinoStrings/index.html#reboot)

However to convert to SafeString you need to pass a SafeString &result as an argument because SafeString prohibits the memory allocation needed to handle a return value.
See the [SafeString library](https://www.forward.com.au/pfod/ArduinoProgramming/SafeString/index.html) tutorial.  The section on [Passing SafeStrings to methods as references and returning SafeString results](https://www.forward.com.au/pfod/ArduinoProgramming/SafeString/index.html#passing)

```
void sendData(SafeString& response , SafeString& command, const int timeout, boolean debug) {
    response = "";  //initialize a String variable named "response". we will use it later.
    Serial.print(command);   //send the AT command to the esp8266 (from ARDUINO to ESP8266).
    long int time = millis(); //get the operating time at this specific moment and save it inside the "time" variable.
    while( (time+timeout) > millis()) //excute only whitin 1 second.
    {      
      while(Serial.available())   //is there any response came from the ESP8266 and saved in the Arduino input buffer?
      {
        char c = Serial.read(); //if yes, read the next character from the input buffer and save it in the "response" String variable.
        response+=c; //append the next character to the response variabl. at the end we will get a string(array of characters) contains the response.
      }  
    }   
    if(debug)  //if the "debug" variable value is TRUE, print the response on the Serial monitor.
    {
      Serial.print(response);
    }    
    return;  //return the String response.
}
```
Then somewhere at the top of your code create the SafeString to hold the response and command
```
createSafeString(response,80);  // good for 80 response
createSafeString(command,80); // good for 80 char cmd
```
Now your real problem is outputing the errors since you are using Serial for the ESP AT coms.
Two things you can do:-
i) set up a SoftwareSerial to send the errors to   SafeString::setOutput(softwareSerial) and connect that to your computer (via TTL-usb cable OR another UNO (softwareSerial in Serial out to your computer)
ii) Turn on the UNO led on SafeString errors
i.e.
```
void loop() {
    if (SafeString::errorDetected()) {
       digitalWrite(ledPin,HIGH);
    }
 // . . . 
```

Attigliuzzo, 2021-07-23:
Thanks for your tricks and your library (especially the GPS example, i learned a lot).

I have already implemented a reboot of my esp8266 every 24 hours using AT commands, i hope this solve my issues but i'll read your links.
My project uses an Arduino UNO but at home i test the code with a mega pro with its multiple serial ports to redirect the debug.
Thanks again

closed 2021-07-23 by Attigliuzzo