PowerBroker2/SafeString #50: SafeString from Arduino String?

ModischFabrications, 2021-12-28:
The documentation shows a few ways to create SafeStrings from raw inputs, but none from Arduino Strings. I would like to port over some of my code, but existing libraries all use generic String. 

I guess a sidecut through String.c_str could work, but it seems risky regarding automatic updates. Could you extend the readme, ideally adding even an example for it?

ModischFabrications, 2021-12-28:
Okay, found it, it's described in https://www.forward.com.au/pfod/ArduinoProgramming/SafeString/index.html#converting ,  but difficult to find. Feel free to close this issue, seems like it's documented already. Still, this conversion was my initial contact point and it probably will for other newcomers as well, it might be sensible to link it somewhere as a "quick migration" guide.

```
    createSafeString(sMsg, 80);
    sMsg = msg.text.c_str();
```

drmpf, 2021-12-29:
Thanks for that let me put a link higher up.

drmpf, 2021-12-29:
Updated https://www.forward.com.au/pfod/ArduinoProgramming/SafeString/index.html
with a bold note under Quick Start.
Please comment back here if you think more is needed, or just close this case.  Thanks

closed 2021-12-29 by ModischFabrications