PowerBroker2/SafeString #66: SafeString causes 'warning "F" redefined' on Teensy 3.2

daanvV, 2023-07-13:
Hello, 

When compiling a for a Teensy 3.2 I get the following warning: 

```
In file included from R:\XXXXX\teensy_test_safeString\teensy_test_safeString.ino:1:0:
c:\Users\XXXXX\Documents\Arduino\libraries\SafeString\src/SafeString.h:138:0: warning: "F" redefined
 #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
 ^
In file included from C:\Users\XXXX\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.57.2\cores\teensy3/Print.h:38:0,
                 from C:\Users\XXX\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.57.2\cores\teensy3/Stream.h:24,
                 from C:\Users\XXXXX\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.57.2\cores\teensy3/HardwareSerial.h:274,
                 from C:\Users\XXX\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.57.2\cores\teensy3/WProgram.h:46,
                 from C:\Users\XXXXX\AppData\Local\Temp\arduino\sketches\6549A712D2E471CF982AC152A4C6C0CD\pch\Arduino.h:6:
C:\Users\XXXXX\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.57.2\cores\teensy3/WString.h:45:0: note: this is the location of the previous definition
 #define F(string_literal) ((const __FlashStringHelper *)(string_literal))
 ^
```

In Teensy 3.2 the `F()` macro does nothing, however it is still defined in source code for the code to be compatible with AVR based Teensy boards. So for Teensy 3.2 boards (and some of the other Teensy ARM boards), would it make sense to disable the `#define F()` in the `SafeString.h` library? 

Many thanks,

PowerBroker2, 2023-07-13:
@drmpf

drmpf, 2023-07-14:
Thank you for your interest in SafeString.
The warning is benign and arises in V4.1.26 and later.
the
#ifndef F
in SafeString.h which previously suppressed this warning was was commented out the handle the current ESP32 code base

Just ignore the warning, i will 'fix' this in the docs by adding a note to the SafeString webpage.

@PowerBroker2 This can be closed. Thanks

closed 2023-07-14 by PowerBroker2