PowerBroker2/SafeString #68: Fix warnings about `F` redefinition on ESP32 (and elsewhere?)

egnor, 2023-10-03:
Compiling SafeString on many architectures generates a lot of warnings about redefining the `F` macro.

From [the code](https://github.com/PowerBroker2/SafeString/blob/cd3c271685a6d5dbb7a33b526791a4f476149add/src/SafeString.h#L137):
```
//#ifndef F  // Arduino IDE 2+ adds namespace around this macro so need to define it here also
// ESP32 defines a macro F(s) (s), so will get a redefinition warning which can be ignored
#define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
//#endif
```

IMVHO, a comment saying "this warning can be ignored" is not a great solution-- many of us like our compilation to be warning-clean as a matter of good practice, and having to somehow remember that _this_ (repeated) warning is OK makes it a lot easier for other warnings to slip through the cracks.

I don't understand the comment "Arduino IDE 2+ adds namespace around this macro". The macro is part of the Arduino runtime, not part of the IDE, macros can't be namespaced in any version of C++, and I don't see any namespaces involved in [the AVR runtime's definition of `F(...)`](https://github.com/arduino/ArduinoCore-avr/blob/63092126a406402022f943ac048fa195ed7e944b/cores/arduino/WString.h#L38), and I can't find that do-nothing definition for ESP32 or any other platform?? Probably I misunderstand something??

I'm not actually sure why this definition has to be there at all?

PowerBroker2, 2023-10-03:
@drmpf

egnor, 2023-10-03:
(By the way, I'm happy to make a PR if desired, just let me know what solution is preferred!)

drmpf, 2023-10-03:
The Arduino SAMD Boads install defines 
class __FlashStringHelper;
inside the arduino namespace in String.h
No warnings when compiling for one of those boards on Default compiler warnings with V4.1.28

Looks like if the SafeStringNameSpaceStart.h etc files have suitable #define checks there is no issue removing the F(string) define.

So I will remove

```
//#ifndef F  // Arduino IDE 2+ adds namespace around this macro so need to define it here also
// ESP32 defines a macro F(s) (s), so will get a redefinition warning which can be ignored
#define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
//#endif

```
in the next release and see if anyone complains about things breaking on one of many boards I don't have on hand to test against.
In the mean time you can just try deleting those lines yourself to check for your own boards.

closed 2024-05-22 by drmpf