PowerBroker2/SafeString #71: Compile Error

Eddiiie, 2023-11-02:
Hi,
I am trying to use .replace() and I am getting an error.  I must not be getting the syntax correct.   How do I pass a single hex character in the replace() function?

Message1Text.replace("%R", '0x14');

I want to replace any occurrence of %R  with the the hex value of 0x14

If I put 0x14 in quotes -   "0x14"  it compiles OK but that is not what I want.  I want to replace %R with the single byte 0x14  .. ?

" error: no matching function for call to 'replace(const char [3], int)'       Message1Text.replace("%R", '0x14');" 

What am I doing wrong?

drmpf, 2023-11-02:
Sorry SafeString has a 
replace(char, const char*)  version to replace a char with a string but not the other way round 
so you have to use the replace(const char*, const char*)  version, replace string with a string of one char.
Your problem is expressing 0x14 as a char in a string
use "\x14"
so you statement becomes
Message1Text.replace("%R", "\x14");

Just as a passing comment, 0x14 is un-unusual char DC4 (device control 4)
Did you mean to use octal 014  "\014" which is FF form feed  also "\x0C"?

Eddiiie, 2023-11-02:
Thank you, it works now!

Yes, 0x14 is correct.    I've been studying packet captures.  I don't fully understand the commands but got enough of them to make the display do some cool things.   

In this case, 0x14 is Reverse text.  There is also 0x12 for underline, 0x16 for Flash, 0x10 cancels reverse/flashing text.. 0x18 cancels it, too... 

Thank you!

closed 2023-11-02 by PowerBroker2