PowerBroker2/SafeString #73: SafeString won't compile on SOME SAMD21 boards, works on others

mgrusin, 2024-05-22:
Hi, I've had this problem for a while but ran into it again today so finally decided to ask:

Safestring always compiles properly for the Arduino MKR Zero board definition.

It has fatal compilation errors on (so far), Sparkfun's SAMD21 boards ("SAMD21 Dev Breakout"), and Adafruit's Feather M0 ("Basic").

These boards all use the same MPU and are electrically similar so this "must" be coming from something in the board support package(s). (Quotes because I know errors come from strange places...)

I'm using the latest Arduino IDE (2.3.2) and Safestring library (4.1.31).

The first error when compiling is:

`In file included from C:\Users\mgrus\Documents\Arduino\VGM-A390-1_V10\VGM-A390-1_V10.ino:99:
c:\Users\mgrus\Documents\Arduino\libraries\SafeString\src/SafeStringReader.h:73:44: error: expected class-name before '{' token
   73 | class SafeStringReader : public SafeString {
      |                                            ^`

I can sort of work around this by forcing the MKR Zero definition, but then I have to manually figure out and translate the Arduino port numbers because people wire their boards differently. This is inconvenient but works.

Any quick suggestions / something I'm doing wrong? I really like the Safestring library and would love including it to be painless.

Thanks so much! -Mike

drmpf, 2024-05-23:
The problem is **namespace arduino**.
The Arduino boards need this, while Adafruit and Sparkfun SAMD board support don't need it and the compile fails if it is used.

The three files 
_SafeStringNameSpace.h
SafeStringNameSpaceEnd.h
SafeStringNameSpaceStart.h_
handle arduino namespace usage.

BUT in the current release ALL boards that define ARDUINO_ARCH_SAMD for compiles are assumed to use arduino namespace
Adafruit and Sparkfun SAMD boards have this define and so are failing.

Adafruit has a ARDUINO_SAMD_ADAFRUIT define which I can use to fix this for Adafruit boards
BUT Sparkfun has not supplied any unique define to test so I can tell SafeString is being compiled for a Sparkfun SAMD board as opposed to an Arduino SAMD board

I will fix the code to look for ARDUINO_SAMD_ADAFRUIT but no 'general' fix available, at the moment, for Sparkfun SAMD boards.

**_Your immediate fix for non-Arduino SAMD boards, is to remove all the contents of those three .h file.  I.e. just have three empty files._**

In the longer term, after the above fix is made, either Sparkfun will provide a define to use 
OR 
you will have to add -DARDUINO_SAMD_ADAFRUIT  to the compile patterns in Sparkfun's  platform.txt file for the SAMD boards 
OR
 have special version of the SaftString library for Sparkfun SAMD boards with empty SaftStringNameSpace....h files

drmpf, 2024-05-24:
V4.1.32 implements fix for Adafruit SAMD boards

closed 2024-06-26 by drmpf