PowerBroker2/SafeString #76: class SafeString' has no member named 'setLength'; did you mean 'length'?

StefanL38, 2024-08-17:
Hi Mathew,

I tried to compile the example-code
C:\Users\....\SafeString\examples\SafeString_Tests\SafeStringLengthTrim\SafeStringLengthTrim.ino

whitch uses a function **setLength()**

but get an error class SafeString' has no member named 'setLength'; 
So I looked up the reference and there is no function setLength.

Was this function removed?
If so the example should not use this function anymore

my compile-log.

Arduino: 1.8.19 (Windows 10), TD: 1.58, Board: "Arduino Uno"

C:\Arduino-Pure-Portable\arduino-1.8.19\arduino-builder -dump-prefs -logger=machine -hardware C:\Arduino-Pure-Portable\arduino-1.8.19\hardware -hardware C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages -tools C:\Arduino-Pure-Portable\arduino-1.8.19\tools-builder -tools C:\Arduino-Pure-Portable\arduino-1.8.19\hardware\tools\avr -tools C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages -libraries C:\Arduino-Pure-Portable\arduino-1.8.19\portable\sketchbook\libraries -fqbn=arduino:avr:uno -vid-pid=1A86_7523 -ide-version=10819 -build-path C:\Users\dipl-\AppData\Local\Temp\arduino_build_546758 -warnings=all -build-cache C:\Users\dipl-\AppData\Local\Temp\arduino_cache_695065 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\dipl-\AppData\Local\Temp\7zO0D1730B5\SafeStringLengthTrim\SafeStringLengthTrim.ino

C:\Arduino-Pure-Portable\arduino-1.8.19\arduino-builder -compile -logger=machine -hardware C:\Arduino-Pure-Portable\arduino-1.8.19\hardware -hardware C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages -tools C:\Arduino-Pure-Portable\arduino-1.8.19\tools-builder -tools C:\Arduino-Pure-Portable\arduino-1.8.19\hardware\tools\avr -tools C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages -libraries C:\Arduino-Pure-Portable\arduino-1.8.19\portable\sketchbook\libraries -fqbn=arduino:avr:uno -vid-pid=1A86_7523 -ide-version=10819 -build-path C:\Users\dipl-\AppData\Local\Temp\arduino_build_546758 -warnings=all -build-cache C:\Users\dipl-\AppData\Local\Temp\arduino_cache_695065 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\dipl-\AppData\Local\Temp\7zO0D1730B5\SafeStringLengthTrim\SafeStringLengthTrim.ino

Using board 'uno' from platform in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.6

Using core 'arduino' from platform in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.6

Detecting libraries used...

"C:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "C:\\Users\\dipl-\\AppData\\Local\\Temp\\arduino_build_546758\\sketch\\SafeStringLengthTrim.ino.cpp" -o nul

Alternatives for SafeString.h: [SafeString@4.1.29 arduino_485041@4.1.29]

ResolveLibrary(SafeString.h)

  -> candidates: [SafeString@4.1.29 arduino_485041@4.1.29]

"C:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\sketchbook\\libraries\\SafeString\\src" "C:\\Users\\dipl-\\AppData\\Local\\Temp\\arduino_build_546758\\sketch\\SafeStringLengthTrim.ino.cpp" -o nul

"C:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\sketchbook\\libraries\\SafeString\\src" "C:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\sketchbook\\libraries\\SafeString\\src\\BufferedInput.cpp" -o nul

"C:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\sketchbook\\libraries\\SafeString\\src" "C:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\sketchbook\\libraries\\SafeString\\src\\BufferedOutput.cpp" -o nul

"C:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "-IC:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\sketchbook\\libraries\\SafeString\\src" "C:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\sketchbook\\libraries\\SafeString\\src\\PinFlasher.cpp" -o nul

"C:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe [... cut ...]
             ^~~~~~~~~

             length

Multiple libraries were found for "SafeString.h"

 Used: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\sketchbook\libraries\SafeString

 Not used: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\sketchbook\libraries\arduino_485041

Using library SafeString at version 4.1.29 in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\sketchbook\libraries\SafeString 

exit status 1

'class SafeString' has no member named 'setLength'; did you mean 'length'?

drmpf, 2024-08-17:
You are correct that function was dropped and the example should be remove, which I will do in the next release.  substring( ) can be used to do that function.

drmpf, 2024-08-20:
Fixed in V4.1.34

closed 2024-08-20 by drmpf