PowerBroker2/SafeString #86: BufferedOutput: `delay()` in `nextByteOut()` (when `!streamPtr`) and in `connect()` causes reentrancy

dakotahorstman, 2025-08-25:
I occasionally use the "hidden" function `yield()` in Arduino core to maintain certain tasks like watchdogs during `delay()` timeouts. I would like to add stream buffer management to this, but I am unable to due to a reentrancy condition.

MRE:
```cpp
#include <Arduino.h>
#include <SafeString.h>
createBufferedOutput(out, 128, DROP_UNTIL_EMPTY);

void yield() {
  // emulate my app that pumps output from yield()
  out.nextByteOut();
}

void setup() {
  Serial.begin(115200);
  // out.connect(...) internally calls delay(...) (and nextByteOut() calls delay if !streamPtr)
  // This re-enters yield() -> nextByteOut() before connect has finished.
  out.connect(Serial);
  out.print("hello");      // never reached / may crash/reset
}

void loop() {
  out.nextByteOut();
}
```

`cores/arduino/wiring_time.c`:
```cpp
void delay(uint32_t ms)
{
  if (ms != 0) {
    uint32_t start = getCurrentMillis();
    do {
      yield();  // reentrancy happens here
    } while (getCurrentMillis() - start < ms);
  }
}
```

This can easily be solved by me guarding `nextByteOut()`, but I still wanted to let you know. In the event you wanted to solve this from SafeString's side and make the library just that little bit easier to use.

drmpf, 2025-08-25:
Try this mod to nextByteOut and see if that solves you issue

```
static volatile bool inNextByteOut = false; // lock to prevent recursive calls to nextByteOut

// NOTE nextByteOut will block if baudRate is set higher then actual i/o baudrate
void BufferedOutput::nextByteOut() {
  if (!streamPtr) {
    SafeString::Output.println();
    SafeString::Output.println(F("BufferedOutput Error: need to call connect(..) first in setup()"));
    SafeString::Output.println();
    SafeString::Output.flush();
  //  delay(5000);
    return;
  }
  if (inNextByteOut) {
    return;
  }
  inNextByteOut = true;
  
  if (mode != DROP_UNTIL_EMPTY) {
    waitForEmpty = false; // always skips a lot of the code below
  }

  int serialAvail = internalStreamAvailableForWrite();
  if (rb_available() == 0) { // nothing to send from ringBuffer
    if (txBufferSize != 0) {
      if (serialAvail >= txBufferSize) { // works is txBufferSize == 0 also
        txBufferSize = serialAvail;
        waitForEmpty = false; // both buffers empty
      }
    } else { // no txBuffer
      waitForEmpty = false; // ringBuffer empty
      // no txBuffer so using baudrate to release bytes instead of availableForWrite()
      sendTimerStart = micros(); // restart baudrate release timer
    }
    inNextByteOut = false;
    return; // nothing to release
  }

  // here have something to release
  //  serialAvail set above
  bool serialBytesWritten = false;
  if (txBufferSize != 0) { // common case use internalStreamAvailableForWrite() to throttle output
    // check if space available and fill from ringBuffer  some boards return 0 for availableForWrite
    if (serialAvail > 0) { // have at least 1 space have already adjusted for ESP32 bug in internalStreamAvailableForWrite
      int toWrite = serialAvail;
      int rbAvail = rb_available();
      if (rbAvail < toWrite) { // limit by number of ringBuffer chars
        toWrite = rbAvail;
      }
      serialBytesWritten = (toWrite > 0); //set once here
      for (int i = 0; i < toWrite; i++) {
        uint8_t b = (uint8_t)rb_read();
        if (b) {
          streamPtr->write(b);
        }
        // else  skip protect bytes '\0'
      }
    }
    // here have either filled txBuffer OR emptied rb_buffer
    // if serialBytesWritten then wrote to txBuffer
    if ((!waitForEmpty) || serialBytesWritten || rb_available() ) { // if just wrote somthing or still have something to write then => waitForEmpty unchanged,
      //  also if waitForEmpty false no need to check (wrong mode)
      inNextByteOut = false;
      return; // not empty
    }
    // here waitForEmpty true && noBytesWritten to txBuffer && rb_buffer empty

    // else need to check for waitForEmpty -> false
    // else waitForEmpty && !serialBytesWritten && !rb has bytes (i.e.ringBuffer is empty)
    // using Serial tx buffer
    int avail = internalStreamAvailableForWrite();  // bytesToBeSent() updates txBufferSize
    if (avail >= txBufferSize) { // empty
      waitForEmpty = false;
    }
    inNextByteOut = false;
    return;
  } // else no txBuffer release on timer

  // txBufferSize == 0 so use timer to throttle output
  // sendTimerStart will have been set above

  unsigned long us = micros();
  // micros() has 8us resolution on 8Mhz systems, 4us on 16Mhz system
  // NOTE throw away any excess of (us - sendTimerStart) > us_perByte
  // output will be slower then specified
  if ((us - sendTimerStart) < us_perByte) {
    inNextByteOut = false;
    return; // nothing to do not time to release next byte
  }
  // else send next byte
  sendTimerStart = us; //releasing next byte, restart timer
  uint8_t b = (uint8_t)rb_read();
  if (b) {
    streamPtr->write(b);  // may block if set baudRate higher then actual I/O baud rate
  }
  // else skip protect bytes '\0' This also skips this release baud rate interval
  if (rb_available() == 0) {
    waitForEmpty = false;
  }
  inNextByteOut = false;
}

```

dakotahorstman, 2025-08-28:
Yep, that seems to have fixed it. Thank you!

drmpf, 2025-08-30:
Ok I will put the in the next release.